nikivanov/mural #16: GCODE

rwlloyd, 2025-03-22:
Hi. Just watched your video and thought. 'Damn, this guy has done all the things I couldn't be bothered to do when I made a wall plotter a few years ago'. I approached from the modded 3D printer direction though.. RAMPS board, [Marlin-polargraph](https://github.com/i-make-robots/Marlin-polargraph) and motors suckered to a window. It was a mess of wire and a pain to set up. Your solution is much neater. Well done!

What might work is to marry the frontend and hardware that you've designed and have another board interpret gcode. The mural board would feed the gcode over serial to the GRBL board (any arduino based or the atmega328). Then 'mural-GRBL' (preconfigured for delta kinematics) handles all of the motor control. 

I think that leveraging and modifying what you already have running and then having mural firmware interpret gcode would be my ideal plotter and I'd start drawing on the big windows at work asap. Your system looks very much simpler and neater than my efforts.   

I guess this is not strictly an issue but I'm going to print a mural when I get the chance. Just for the hardware design I will thank you in advance.

nikivanov, 2025-03-25:
There is already an ESP32 compatible version of Grbl and a newer version, called FluidNC. I'm planning on adapting Mural to work with FluidNC so I can get all the movement goodies that it comes with.

Yash-xoxo, 2025-03-29:
thanks it might me a good upgrade

closed 2025-04-01 by nikivanov