nikivanov/mural #18: Build notes/park servo error

PangeaPrint, 2025-03-25:
First off great work! Couple things:

1. The "Park Servo" button doesn't do anything for me to control the pen holder servo. When plugged in it goes to the in/up position on boot up which works but when attaching the linear motion piece it needs to be put on in a different orientation than the video. It's really not an issue with functionality yet just wanted to add that note if anyone is seeing the same. Properly orienting it allows it to work when adjusting distance later in the prep menu.

2. Here's the wiring for the L/R servos:

4pin side               6 pin side
1-2-3-4---------------4-EMPTY-1-3-EMPTY-2
ex. Black-Green-Blue-Red------------------Red-EMPTY-Black-Blue-EMPTY-Green

[image]

I have the entire unit built/wired/wifi connected but one of the L/R stepper control boards showed up with the voltage adjustment screws busted so it's stuck at low voltage and won't operate under a load (ie it's own weight) without skipping until the new one comes tomorrow. That being said everything is acting correctly during bench testing.

I'm using a generic ESP32 (ESP-WROOM-32 ESP32 ESP-32S Development Board from Amazon) since I had it floating around so that may cause issue as well. I'll update tomorrow if I can get it to draw.

nikivanov, 2025-03-26:
You're absolutely right! When I was reshuffling the screens I forgot to add servo control to the Set Top Distance screen. I'll fix it now.

nikivanov, 2025-03-26:
@PangeaPrint I just uploaded the new firmware - could you try reflashing your ESP32 and let me know if that fixes your servo issue?

PangeaPrint, 2025-03-26:
Yup, that worked! One more question, not sure what the Extend 100mm (e-steps calibration) button is for but I don't see it making any movements.

nikivanov, 2025-03-26:
Yeah that one is broken too - I haven't used it in ages. I wrote it to verify the extrusion length during initial design. I'll fix it at some point.

closed 2025-03-29 by nikivanov