nikivanov/mural #21: Stuck on rasterizing

twofieros, 2025-03-27:
First off, thank you for this project. I had a Scribit but that company is no more, so I just built this project, and it was so much fun and easy to do. Now I'm trying to get it to draw but all the svg files I try don't seem to work. When I select the file, I see a broken image icon initially. Then when I hit preview, it just rasterizes and nothing happens. I'm making the svg myself with silhouette studio. I have everything set to black only. I've made sure the lines are at least one pixel in thickness, and I've tried infill and without, but nothing seems to be working. Am I missing something basic?

nikivanov, 2025-03-27:
Please upload one of your svgs here so I can try to debug it

twofieros, 2025-03-27:
[image]

Just a simple Test spelled out.

I was able to get a downloaded svg file to work of the web so the mural does work just I can't seem to make my own files. What program did you use to create the svg files?

nikivanov, 2025-03-27:
I can reproduce the issue, but I'm not sure why it's happening. Try using Inkscape to create your SVGs. If the preview image appears to be broken, then it's safe to say further processing won't work either.

PangeaPrint, 2025-03-28:
I'm using AI 2025 and saving as an SVG 1.1 if that helps. Your svg works just fine when opened and resaved so it may be the options for exporting/saving as an SVG in Silhouette Studio.

[image]

[image]

twofieros, 2025-03-28:
Thank you both for this. I will use Inkscape to save my files from now on.

closed 2025-03-29 by nikivanov