nikivanov/mural #22: 3d Files for byj48 Version (as seen in your Video) with unipolar to bipolar mod and MKS-DLC32

flow1986, 2025-03-27:
Hi,
very nice Project! would it be possible for you to upload the BYJ48 Stepper Version Stls?

You can use these motors with an Bipolar mod and some Bipolar Motor Drivers for more torque and less lost steps.

i want to rebuild your Mural Walldrawer with an mks-dlc32 (https://github.com/makerbase-mks/MKS-DLC32)

because it would be even cheaper and with the mks-dlc32 everything is on board (esp32, stepper driver, power supply)

i have all parts at home, because i wanted to build a wallplotter with FluidNC, but didn´t have the time yet...

thanks Flo

nikivanov, 2025-03-27:
Sure, but I wont be able to provide any support for it, and the existing firmware won't work on it. If that sounds good I can upload the STEP file

flow1986, 2025-03-27:
That Sounds perfect for me!
Thank you

nikivanov, 2025-03-27:
https://github.com/nikivanov/mural/blob/28byj-48/STLs/prototype%20v8.step

closed 2025-03-27 by nikivanov