nikivanov/mural #25: Change pins in software

ModestoHagney, 2025-03-29:
Would it be possible to add the ability to change the pins, and maybe some other settings like speed and servo positions in software?

The setpper board I'm working with uses pinouts that are different to the ones in the code.
I did manage to recompile and change the pins but I had heaps of issues getting it to compile.

nikivanov, 2025-03-29:
Hi! I'm not going to be adding support for other boards to Mural, since its hardware and software are interdependent. You're welcome to fork the repo and add these abilities if you want to make the software more generic. It sounds like you got the firmware to recompile, so you're already half way there.

ModestoHagney, 2025-03-30:
Fair enough. I had a hell of a time getting it to compile, something to do with WifiManager.git but I managed to get it on there in the end.

closed 2025-03-30 by nikivanov