nikivanov/mural #26: Coordinate conversion code

CIRCUIT-RAPTOR, 2025-03-29:
Hi. I'm trying to implement a similar plotter using Raspberry pi Zeo 2W. I would like to know how the svg file uploaded is converted to coordinates for the stepper motor. Is there a specific code for doing this operation? I just want to know how the svg is converted to the final form as seen in the drawings.

nikivanov, 2025-03-29:
The code runs in your browser, with some of it running in the web worker. I rasterize and re-vectorize SVGs to account for this like stroke width. 

Here's the vectorization call to the web worker: https://github.com/nikivanov/mural/blob/4682db1fa4d548cf25d1176ae29ae6e4425b6ff8/data/www/main.js#L167

Here's the call to the web worker that converts the re-vectorized SVG into a list of commands for Mural: https://github.com/nikivanov/mural/blob/4682db1fa4d548cf25d1176ae29ae6e4425b6ff8/data/www/main.js#L206

Web worker code is written in typescript and then webpacked. Here's the entry point of rendering the SVG into a list of commands: https://github.com/nikivanov/mural/blob/4682db1fa4d548cf25d1176ae29ae6e4425b6ff8/tsc/src/main.ts#L39

CIRCUIT-RAPTOR, 2025-04-03:
I was able to replicate the outline drawing function. But I'm not able to achieve the infill/crosshatch pattern in those dark areas. Any advice on how to make this work? Also, I might switch to an ESP-32 if this turns out to be unsuccessful, because I need this to be completed within 2 weeks. I'm trying to use things available at my house, that is why I went for the pi, 28byj-48 stepper motors and ULN2003 driver. If I had to use the ESP-32, would the program flashed in it via the mural website be able to run these 28byj-48 stepper motors? I saw them on your video that it was used in the first iteration. 
Thanks

nikivanov, 2025-04-03:
The SVG processor will shade in any paths that have a non-white non-transparent infill. The density of the crosshatch infill is governed by this property: https://github.com/nikivanov/mural/blob/4682db1fa4d548cf25d1176ae29ae6e4425b6ff8/tsc/src/main.ts#L97

You won't be able to use the current firmware with 28byj-48 - they are driven completely differently from the steppers that I'm using.

CIRCUIT-RAPTOR, 2025-04-05:
I'm still not able to successfully create an infill code. So, I'm now using an esp-32 to flash the firmware through mural website. But I'm getting stuck at erasing flash. I've read a solved issue about the same and saw that he flashed it via VS Code in the end. Is there any fix?

[image]

nikivanov, 2025-04-05:
Try some of the suggestions in this thread and let me know if it helps: https://github.com/nikivanov/mural/issues/30

CIRCUIT-RAPTOR, 2025-04-05:
I've also tried those. I don't think my problem is with connecting, as from the picture it is clear that the device is connected and is able to recognize it. I tried to flash it via VS Code with Platform IO. But I'm getting errors while building:

Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.10.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- WiFiManager @ 2.0.17+sha.32655b7
|-- ESPAsyncWebServer @ 3.7.4+sha.25845e7
|-- ESP32Servo @ 3.0.6+sha.e697ea7
|-- Adafruit SSD1306 @ 2.5.13+sha.78d6a13
|-- AccelStepper @ 1.64.0
|-- ArduinoJson @ 5.13.4
|-- SPIFFS @ 2.0.0
|-- AsyncTCP @ 3.3.8
|-- ESPmDNS @ 2.0.0
|-- FS @ 2.0.0
|-- Wire @ 2.0.0
Building in release mode
Transpiling TS code
rm data/www/worker/* || true
'rm' is not recognized as an internal or external command,
operable program or batch file.
'true' is not recognized as an internal or external command,
operable program or batch file.
*** Error 1
npm run build
'npm' is not recognized as an internal or external command,
operable program or batch file.
*** Error 1
cp dist_packed/main.js ../data/www/worker/worker.js
'cp' is not recognized as an internal or external command,
operable program or batch file.
*** Error 1
Retrieving maximum program size .pio\build\esp32dev\firmware.elf
Checking size .pio\build\esp32dev\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  15.2% (used 49796 bytes from 327680 bytes)
Flash: [========  ]  81.6% (used 1069577 bytes from 1310720 bytes)
============================================================== [SUCCESS] Took 8.94 seconds ==============================================================

I tried uploading. Even though it shows successful upload, I am not able to find the esp hotspot. I tried monitoring it and it shows a SPIFFS mounting error. 

I'll provide an update if I find a fix.

nikivanov, 2025-04-05:
What OS are you running? rm and cp are Linux terminal commands.

I'll provide a script you can run to flash the firmware locally.

CIRCUIT-RAPTOR, 2025-04-05:
I'm on windows 11.

nikivanov, 2025-04-05:
I'd recommend you use wsl to compile the firmware yourself. I'll also
provide a python script to flash the firmware locally

nikivanov, 2025-04-06:
@CIRCUIT-RAPTOR I wonder if you have the same error in your browser console as here: https://github.com/nikivanov/mural/issues/30#issuecomment-2781172557

CIRCUIT-RAPTOR, 2025-04-08:
Thank you. Finally flashed it via the website you provided.
https://getmural.me/firmware_flasher_slow.html

closed 2025-04-08 by CIRCUIT-RAPTOR