nikivanov/mural #28: GT2 drive belt slipping

litxaohu, 2025-03-31:
Thanks to the author for open sourcing the project, I have successfully reproduced it. 

However, the buckle 3D file with the belt fixed does not seem to be suitable. It is difficult to fix and causes the belt to fall off. I tried the 4mm and 2mm 3D printer extrusion heads, but the results are similar.

I designed a new fixed clap, which uses the friction between the belts to fix, and the precision requirement of the printer is relatively low. I hope the author can answer my related question. There are some differences between the size of the clasp and the original one, will it affect the printing accuracy?

[image]

Thanks again to the author for the project, it's a great design.
I have attached all my [printed documents here](https://github.com/litxaohu/mural/blob/main/GT2_v2.stl).

[image]

nikivanov, 2025-03-31:
Do you mean that the belt does not stay in the hanger and falls out? If so, you can try the trick I suggest in the video - cut the end of the belt so it has a frayed end and that little extra material would make it sit better in the hanger belt trough.

If you're using your own hangers, the accuracy should not be affected as long as you home the device by placing the hanger loops on the homing bolts.

litxaohu, 2025-04-01:
I checked your video carefully and I found that the belt I bought may be a little different, when I cut off the end of the belt, it will not become rough and therefore difficult to secure.

nikivanov, 2025-04-01:
I see. Your hanger should work just fine then. Worst case you can just hot glue the belts into the hangers.

closed 2025-04-03 by nikivanov