nikivanov/mural #30: Firmware Flasher

toxict0ast, 2025-04-02:
Very cool project! I'm excited to try it. Sadly, amazon sent the wrong pully, so I'm waiting on that. 

I can't seem to get the webflasher to work. 

1. I installed the CP210x_Windows_Drivers
2. Held down boot, pluged in the USB, let go of boot button.
3. Clicked "Flash firmware" and I just get stuck here

esptool.js
Serial port WebSerial VendorID 0x10c4 ProductID 0xea60
Connecting...

Detecting chip type...ESP32
Chip is ESP32-D0WD-V3 (revision 3)
Features: Wi-Fi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: d8:13:2a:2f:4a:08
Uploading stub...
Running stub...
Stub running...
Changing baudrate to 460800
Changed
Erasing flash (this may take a while)...

I have tried a few USB cord, edge/chrome, and 2 different ESP32 boards. I am able to flash other things to it, like ESPhome as a test. Any ideas what am I doing wrong?

nikivanov, 2025-04-02:
I've never seen it crash there. No idea, sorry. Please let me know if you figure it out. Try checking chrome devtools console log for errors.

nikivanov, 2025-04-02:
Comparing the output you posted above to the one I get from my board, your board's chip is ESP32-D0WD-V3 (revision 3) and mine is ESP32-D0WDQ6 (revision 1). Unless you're on a super tight budget, I suggest buying the ESP32 from the Amazon link from the BOM and seeing if that resolves your issue.

Also, another person was telling me about their problems with flashing their controller and they found a somewhat surprising solution - try it first before buying another set of boards?

[image]

twofieros, 2025-04-02:
I had similar issues with Chrome on my Mac when trying to flash. To get it to work I held down both buttons when plugging in. Release them one at a time with about 5 seconds between plugging in and releasing. Release the boot button last.

toxict0ast, 2025-04-02:
@nikivanov  Thanks for replying! I will try you and @twofieros  suggestions. 

I did buy the one you linked on amazon. But I wouldn't be surprised if amazon made a mistake. The also sent me the wrong pulleys. 

[image]

[image]

nikivanov, 2025-04-02:
This definitely looks like the correct board, just a newer revision of mine. Let me know any of the suggestions above make it work for you!

grantmatson, 2025-04-06:
I'm running into the same issue.  My board is the same ESP32-D0WD-V3 (revision 3), which was purchased directly from the BOM.  I've tried all the suggestions in this thread and it still freezes on the "Erasing Flash (this may take a while)...".  

The console has this error:
```
bundle.js:2  Uncaught (in promise) Error: Invalid head of packet (0x78): Possible serial noise or corruption.
    at de.read (bundle.js:2:50152)
    at async Pe.readPacket (bundle.js:2:58998)
    at async Pe.checkCommand (bundle.js:2:64609)
    at async Pe.eraseFlash (bundle.js:2:70652)
    at async Pe.writeFlash (bundle.js:2:75155)
    at async HTMLButtonElement.<anonymous> (firmware_flasher.html:438:13)
```

nikivanov, 2025-04-06:
Thanks for the stacktrace! The more immediate plan is to provide a python script that flashes the firmware, but hopefully I can figure out why it works for some but not the others.

nikivanov, 2025-04-06:
A quick google search suggests two things:

-  Per https://github.com/espressif/arduino-esp32/issues/1253#issuecomment-435420119 try holding the BOOT button throughout the whole process
-  Per https://github.com/espressif/arduino-esp32/issues/1253#issuecomment-487285571 I could try to turn down the baudrate to 115200 here: https://github.com/espressif/arduino-esp32/issues/1253#issuecomment-487285571 . If you know how to do it in DevTools, try changing this value in your browser and then go through the flashing process

grantmatson, 2025-04-06:
I switched to my Macbook and it ran without any issues. I'm probably going to make a second Mural since I have all the parts for it. I'll try your suggestions on my deckstop (Windows 11), which was the system having the error.

CIRCUIT-RAPTOR, 2025-04-06:
I was able to successfully flash using WSL. But when I monitor it, I keep getting this block of output repeatedly:

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
ets Jul 29 2019 12:21:46

I tried flashing a blink program using Arduino IDE and it worked well. Then I used an online ESP flashing tool. In this tool I was able to successfully erase and flash programs (an example program that lets you connect to Wi-Fi), and it had no trouble running those. But whenever I open the terminal after erasing the flash I get this error repeatedly which is similar to the one before.

[11:06:52]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[11:06:52]invalid header: 0xffffffff
[11:06:52]invalid header: 0xffffffff
[11:06:52]invalid header: 0xffffffff
[11:06:52]invalid header: 0xffffffff
[11:06:52]invalid header: 0xffffffff
[11:06:52]invalid header: 0xffffffff
[11:06:52]invalid header: 0xffffffff
[11:06:52]invalid header: 0xffffffff
[11:06:52]ets Jul 29 2019 12:21:46

nikivanov, 2025-04-06:
@grantmatson did you install the driver on win11?

piousgreek, 2025-04-07:
Greetings, i am having similar issue to others.  i purchased the esp32 from the BOM and it identifies as ESP32-D0WD-V3.  CP210x driver is installed. however the mural web flasher is not working.  The cp210x chip on the board appears to be a knock off but i can successfully flash generic sketches from both Arduino and platformio, 

when i try to build the build the project using platformio, i receive some errors and although it produces a firmware that i can flash, the esp ends up in a bootloop complaining about a missing spiff.

can you release or provide a .bin file so i can flash via esptool or some other program.

nikivanov, 2025-04-08:
Hey everyone!

If you're having trouble flashing the firmware for your controller, first try using this page instead:
https://getmural.me/firmware_flasher_slow.html

It's the same page as the original flasher but the baud rate is turned down to 115200.

If that still doesn't work for you, download this ZIP file and run the command in the readme.txt https://getmural.me/firmware/firmware.zip

Please let me know what works and what doesnt!

piousgreek, 2025-04-08:
the slower baud rate with the web flasher seems to have done the trick for me, thank you.  "Firmware flashed successfully!"

CIRCUIT-RAPTOR, 2025-04-08:
Mine also worked after flashing from https://getmural.me/firmware_flasher_slow.html
I think the problem is with the knockoff chip inside the ESP since I think mine is also a knockoff. Maybe this should be added as an alternative flashed in the mural website. 
Thank you for all the effort.

nikivanov, 2025-04-08:
Excellent! I'm pushing the change to the main firmware_flasher.html now, and deleting _slow.html

closed 2025-04-11 by nikivanov