nikivanov/mural #31: Connect Enable Pins on Stepper motor extension board

bkonings, 2025-04-03:
My Motors were not working somehow.

After measuring everything 10 times, verifying my connections and zooming in a lot in YouTube to make sure (does he connect the enable pins or not?) I dove into the code and found you do assign the enable pins in the code.

After I connect those everything is working as expected.

const auto LEFT_STEP_PIN = 13;
const auto LEFT_DIR_PIN = 12;
const auto LEFT_ENABLE_PIN = 14;

const auto RIGHT_STEP_PIN = 27;
const auto RIGHT_DIR_PIN = 26;
const auto RIGHT_ENABLE_PIN = 25;

So now I wondering if you just forgot to connect them up in the YouTube video?
I do use a different ESP32 board, but I don't think that explains the issue.

Thanks for the project, very interesting.

nikivanov, 2025-04-03:
That's really strange, since the motors should be enabled as their default state, so if the enabled pin is not connected they are never NOT enabled. I did make the plumbing in software for the enabled pins, but I never ended up needing a use case when the motor are not enabled. Are you using TMC2209 drivers? Per spec:
```
ENN 2 DI Enable not input. The power stage becomes switched off (all motor
outputs floating) when this pin becomes driven to a high level.
```
which should mean that motors are enabled by default as long as that pin is pulled low.

bkonings, 2025-04-04:
I agree it should work but definitely doesn't without connecting the enable pin.

I am using these: BIGTREETECH TMC2209 V1.3

(https://nl.aliexpress.com/item/33028050145.html?spm=a2g0o.order_detail.order_detail_item.4.5abe6d76pAjHwg&gatewayAdapt=glo2nld)

ESP32 I am using: ESP32 Dev Kit C V4

As I can connect the enable pin it is fine, just wanted to share my experience ;)

nikivanov, 2025-04-04:
Yep, I appreciate you providing the workaround! Looking at AliExpress link, those are version 1.2 whereas Amazon's are 1.3. Unlikely that it's the culprit, but maybe 1.2 doesn't pull the EN pin low by default? Anyways, I'll close this issue soon and create a section for Troubleshooting, where I'll link to this issue.

Dytoractor, 2025-04-18:
I tried to save money and use my old 2209 v1.2 and I ran into the same issue where my motors were not turning. Thank you Nik for showing me this fix!!! It worked! However, I don't have enough v1.2's for the other two units I am making for my nieces. So I ended up buying four V1.3 drivers. 

Curious, does it hurt to just always put the ENN Pin in? no matter which driver  it is. Like the picture I made

[image]

nikivanov, 2025-04-18:
It shouldn't hurt to have the EN pin be connected. Maybe some variants of tmc2209 has En pin floating voltage be high enough to considered high. I'll add a note about it. Thanks!

Dytoractor, 2025-04-19:
Sorry, one more question!

Where do you find your SVG files? or do you have a forum/ "how to" page on
this?
I tried to make some but they ended up being "pictures". I found some free
svg's online that have gcode when you open them and they work... I found
this border (attached). but none of the other svg's show up in the
preview.

thank you

nikivanov, 2025-04-20:
@Dytoractor there's https://freesvg.org/ , I also had decent luck with just typing in what i want with "svg free" in google. Also, as @AMagill recommended, you can ask an AI generator to "create a vector image of [your prompt], line art" and then have it traced using InkScape or potrace

JesusBoraita, 2025-07-18:
@Dytoractor I had the same problem with the motors not moving with TMC 2209 v2.0 drivers. This schematic saved the day by connecting the enable pin. Thank you all so much for your work and effort.