nikivanov/mural #32: Calibration ?

jp3141, 2025-04-05:
What is required for position calibration ? Obviously you don't pull the cables until they are horizontal -- is there a 'top of print' angle or distance from the support points ?

nikivanov, 2025-04-05:
The belts are placed on homing bolts so the initial belt length is known. After that Mural just assumed no steps are skipped.

nikivanov, 2025-04-05:
Oh I don't think that was your question. Mural's drawing margins are 20% from the top and 20% from each side, taken from the distance between the hangers.

jp3141, 2025-04-06:
Yes. That was my Q. I always skipped that step since the ends of the belts are high and inaccessible 🙃. I understand now...  thanks. John--John Pigott

closed 2025-04-06 by nikivanov