nikivanov/mural #33: Scaling issues during preview

mrdavidgagnon, 2025-04-06:
For any .svg I select, the move and resize screen looks great, but then the preview screen shrinks the image and moves it to the top left corner. The print also comes out small. 

FWIW,Thank you for sharing this incredible project![image]

nikivanov, 2025-04-06:
Could you please attach the original SVG?

mrdavidgagnon, 2025-04-06:
New insight! The problem went away when I switched to my desktop Mac computer running Chrome. It seems there is an issue with desktop and mobile safari.

nikivanov, 2025-04-06:
Good to know! I've done 0 testing on safari, Chrome is currently the only supported browser.

StefanMeGit, 2025-04-22:
It would be incredible helpfull when you are able to tell the approximate dimensions of the drawing instead of the "scale factor". 0.X is absolute useless for me... but 1000mm in width would be a great hint how big the drawing would be on the wall.