nikivanov/mural #34: Distortion Correction Fix!

TheSeanLavery, 2025-04-06:
I made a fork of your repo. It corrects the distortion and has adjustable parameters to tweak the correction.

I also built a python based plotter simulator to understand the distortion, and visualize the corrections.

I did this under an hour from seeing your video. Here is the repo https://github.com/TheSeanLavery/mural/tree/distortion-correction

Not sure if it will work 1st try on a real device, but I think this can solve the issue, and it did compile with out errors.

nikivanov, 2025-04-08:
Thanks for making these changes! The main issue is that I'm not really sure where the distortion is. I need to draw a calibration grid with Mural and then manually draw the geometrically accurate version of the same grid with a big ruler. Once I can quantify the distortion, correcting it should be as easy as applying a matrix to the drawing coordinates.

I initially hoped that somebody would know how to accurately model Mural's movement in code, catenary curves and all, but the general consensus so far is that dewarping is the easiest approach.

TheSeanLavery, 2025-04-08:
I think I may actually be modeling Mural's movement with the python visualizer.

If its not doing what it should be doing. I can see about correcting this an alternate way.

You may not need to even run a calibration with my code here.

I don't have a device yet. So I'm unable to test with real hardware.

nikivanov, 2025-04-09:
You're modeling Mural's movements as Mural perceives them to be - same code as in movement.cpp. Mural's input is (x,y) coordinates and output is belt lengths. The issue is that when Mural accepts desired (x,y) coordinates, it converts them to belt lengths, but the actual (x,y) you end up with on the wall are different from the original desired (x,y) coordinates.

As I mentioned above, there are two ways to approach this: model the movement correctly (which I don't know how to do), or use a dewarping matrix by drawing a grid using Mural and then superimposing the same geometrically-correct grid onto it - the resulting matrix can be used to warp the original coordinates to shoehorn Mural's sort-of-incorrect movement model into drawing geometrically-correct images.