nikivanov/mural #36: [Question] Motors temperature

kivson, 2025-04-14:
Hi, thanks for the amazing project.

I just finished my first draw (10 min duration) and the motors are really hot, 
barely can touch them for a second. Is that normal?

Only thing I did different from the assembly video was that my reference voltage between the adjustment screw and grd was already at 1.1 volts, so did no adjustment there. Any suggestion on how reduce the temperature?

nikivanov, 2025-04-15:
In my experience, motors do get hot to the point where you cant hold them, but that doesnt seem to affect performance. One of the members on the discord server suggested I use UART as the interface method between the microcontroller and the driver - it should eliminate the need for setting VREF too. This wouldn't require any new parts, but would slightly change the wiring.

depthpicker, 2025-04-27:
Hi first of all congratulations and thanks for this cool project!
If I may, I would like to share a few observations on the matter of the heat issue. I built this project and have observed significant thermal differences between the mural firmware and my own esp32 code that I run on the same hardware..

It appears that when running the mural firmware with VREF set to a low value of 0.75V, the motors consistently reached temperatures of approximately 50°C, even during periods of inactivity or when not even mounted against a wall. At 1V they get even hotter. With my own simple esp32 code (just a web interface and buttons to manually drive the motors also using Step Dir ), the motors stay cool to the touch. However, upon reverting to the original mural firmware, the elevated temperature issue immediately returned..

nikivanov, 2025-04-27:
Are you willing to share your code so I can compare?

nikivanov, 2025-05-14:
Motors stay energized so they hold position at all times. If your firmware moves them and then disables them, that doesn't really count

AlexYeryomin, 2026-08-30:
I don't have a thermal camera, so I use a _non-scientific_ method to test a max temperature of the stepper: I use a small neodymium magnet placed on the motor. When it reaches ~80°C, the magnet is getting a bit weaker, and it barely holds at ~100°C. Not much of difference, but one can feel it. While drawing murals, the steppers are getting really hot, reaching above ~80°C for sure, but it seems the temp stays below 100°C all the time (ambient ~23°C). 

The steppers work just fine. But because the case is typically 30–50 °C cooler than the windings, it means the windings temperature is too close to 130 °C (allowed by the spec for Class B insulation). I think, I would install small fans over steppers, just in case.

nikivanov, 2026-08-30:
I'm finishing up version 2.0. I use smaller pulleys and a ball bearing to improve effective motor torque. I run motors at 0.7A now they perform great getting barely warm