nikivanov/mural #37: TMC2209 IO logic 5V instead of 3.3V

analogic, 2025-05-08:
Hi Nik, I don't have exactly the same parts as you recommend as I have almost all the parts in house but in slightly different versions. It happens that I have already fried two esp32s and the steppers are still not moving properly. So I have to dig deeper

I found some discrepancies in the video. Should the TMC2209 extension board be connected to the 3.3V from the ESP32 rather than the 5V source - AFAIK it is connected to VCC_IO and the ESP32 outputs 3.3V?

analogic, 2025-05-09:
I would expect VCC_IO to define the voltage level of the IO logic, so if the ESP32 outputs a lower voltage it could potentially cause problems in some extreme cases...

nikivanov, 2025-05-09:
I see what you're saying, but it seems that powering tmc2209 with 5v and then using 3.3v to control step and for just works, for whatever reason, for quite a lot of people.

For your immediate problem, have you tried https://github.com/nikivanov/mural/issues/31 ?

nikivanov, 2025-05-10:
> Ah, OK. I did not find any IO level limits in the datasheet - so feel free
> to close.
>
> My problem is mostly based on unstable connections due to wobbly wiring (I
> don't think duponts are great for moving environments either), fried chips
> due to mishandling, and ghosts :). I'm in a situation where I can't rely on
> any of these parts. One idea was to order a custom board to get rid of
> problematic connections, but I've found and ordered the FYSETC E4, which
> seems cheap and more than ideal for this purpose. The end result should
> also be rock solid. I'm determined to port your FW to this board (PIO makes
> it quite easy)...

Yeah that looks like it has everything you'd need. Let me know how the port
goes!

CCERocks, 2025-05-14:
Hi @analogic & @nikivanov, my FYSETC E4 board just arrive today & I'm flash with getmural firmware... Unfortunately only RIGHT MOTOR is working. 

Any guide to make the LEFT MOTOR working too?

Anyway @nikivanov would you add option to speed?

nikivanov, 2025-05-14:
You can modify drawing speed here: https://github.com/nikivanov/mural/blob/4682db1fa4d548cf25d1176ae29ae6e4425b6ff8/src/movement.h#L6

As for the board, I don't have one and have never used it so I can't provide support for it. I'll keep the issue open in case @analogic gets it to work

CCERocks, 2025-05-15:
this is your setting for motor pin

const auto LEFT_STEP_PIN = 13;
const auto LEFT_DIR_PIN = 12;
const auto LEFT_ENABLE_PIN = 14;

const auto RIGHT_STEP_PIN = 27;
const auto RIGHT_DIR_PIN = 26;
const auto RIGHT_ENABLE_PIN = 25;

For FYSETC E4 

X-MOTOR(1)  <--- RIGHT MOTOR
X-Step | 27  
X-DIR | 26  
X-EN | 25 

Y-MOTOR(2) <--- LEFT MOTOR ? so should follow this FYSETC info to your movement.h right? 
Y-Step | 33
Y-DIR | 32
Y-EN | 25 
 
Z-MOTOR(3)
Z-Step | 14 
Z-DIR | 12 
Z-EN | 25 

E0-MOTOR(4)
E0-Step | 16
E0-DIR | 17 
E0-EN | 25 

[image]

but which pin for Servo ? using Endstops, pin 34, 35, 15 or FAN pin 13 ?

nikivanov, 2025-05-15:
Yep, if you connect the left motor to the Y motor connector, you just need to use those pins from Fysetc E4 spec

CCERocks, 2025-05-16:
> Yep, if you connect the left motor to the Y motor connector, you just need to use those pins from Fysetc E4 spec

Pin 17 for Servo?

Well I have to redesign the casing to fit the new board. Maybe motor can be put this way... Since nothing at that area.

[image]

nikivanov, 2025-05-17:
You can use any of the pins that support PWM (that's most of them, but depends on the exact esp32 chip type). Pick a pin from the available list that isn't already used and that supports PWM, plug the servo into that and modify the pin number here: https://github.com/nikivanov/mural/blob/main/src/pen.cpp#L49

CCERocks, 2025-05-18:
> You can use any of the pins that support PWM (that's most of them, but depends on the exact esp32 chip type). Pick a pin from the available list that isn't already used and that supports PWM, plug [...]

Noted. Thank you for the info... 

Anyway.. its would be a great upgrade to have option to choose speed either with slider or numeric input or even by percentage anywhere in the MURAL Interface.. or just put it inside setting when user can see how fast the motor spin.. 

😅

nikivanov, 2025-05-18:
I hear you, but I wanted to provide the user with "just works" experience. If the drawing speed can be altered, I need to find the min and the max through testing, and I don't have the time. You're welcome to modify the speed in the firmware and flash it on your Mural. If you can confirm higher speeds without sacrifice in quality, please let me know!

CCERocks, 2025-05-25:
> The motors work flawlessly this time! The servo seems to be a bit of a problem – the board itself only has an AMS1117, which does not have enough power to feed the ESP32 and the servo, so a step-down converter from 12V to 5V needs to be used. The SDA/SCL pins (22 and 21) should work to control the ESP32 (on the 3x4 header close to it). I'll try it later. I am also kind of tempted to change the servo to a stepper with code doing a similar thing.

well.. there isa  small stepper motor 28BYJ-48 have 12V version... maybe can use this one without any conversion ?

analogic, 2025-05-28:
A few notes:
- For the servo, I used a Mini360 step-down converter in a heat-shrink tube, which I connected to the hotbed screw terminal. You need to set pin 4 to HIGH before doing anything else.
- To drive the servo, I use the SCL board pin (pin 22 internally, near the ESP32 board).
- The TMC2209s are meant to be commanded by UART on this board (MS1 and MS2 are used for the address, not for microstepping), so some additional code is required to set the same values for the microsteps; otherwise, one will rotate faster.
- Also, the holding current must be set via UART to prevent the motor from getting unnecessarily hot.

CCERocks, 2025-05-30:
> A few notes:
> - For the servo, I used a Mini360 step-down converter in a heat-shrink tube, which I connected to the hotbed screw terminal. You need to set pin 4 to HIGH before doing anything else.
> - [...]

Would you mind sharing your source code?... I can share you my Mural case design that fit this FYSETC E4 board... 😂 Notes.. it's different.

nikivanov, 2025-05-30:
We're talking on github - the source code is all here :) What are you looking for specifically? If it's the STEP file for Mural, it's here: https://github.com/nikivanov/mural/blob/main/STLs/Mural.step

CCERocks, 2025-05-31:
> ~~We're talking on github - the source code is all here. What are you looking for specifically? If it's the STEP file for Mural, it's here: https://github.com/nikivanov/mural/blob/main/STLs/Mural.step~~
> 
> Ah, sorry, I think you were asking @analogic 

LoL... It's ok... I will eventually share the design of your Mural that use FYSTEC board if I get the code correctly. 😀

analogic, 2025-06-04:
Ok, now I've test code for the board and working motors. I'll try to implement to mural code later

Test code:
```
#include <Arduino.h>
#include <TMCStepper.h>
#include <ESP32Servo.h>
#include <AccelStepper.h>

#define LED_BUILTIN 2 // blinking with FAN led

#define SERIAL_RX     21    // PDN_UART → RX2
#define SERIAL_TX     22    // PDN_UART → TX2

#define SERVO_PWR_PIN 4

#define EN_PIN 25

#define STEP_PIN_1 27 // Motor 1 STEP pin
#define DIR_PIN_1 26  // Motor 1 DIR pin

#define STEP_PIN_3 33 // Motor 3 STEP pin
#define DIR_PIN_3 32  // Motor 3 DIR pin

#define SERVO_PIN 15 // Servo motor pin

#define R_SENSE 0.22f

Servo servo;

HardwareSerial tmcSerial(2);

TMC2209Stepper drv1(&tmcSerial, R_SENSE, 1); 
TMC2209Stepper drv3(&tmcSerial, R_SENSE, 3);

AccelStepper stepper1(AccelStepper::DRIVER, STEP_PIN_1, DIR_PIN_1);
AccelStepper stepper3(AccelStepper::DRIVER, STEP_PIN_3, DIR_PIN_3);

void fatalError()
{
    Serial.println("Fatal error occurred. Stopping execution.");
    pinMode(LED_BUILTIN, OUTPUT);
    while (1) {
        digitalWrite(LED_BUILTIN, HIGH);
        delay(500);
        digitalWrite(LED_BUILTIN, LOW);
        delay(500);
    }
}

void setup()
{
    Serial.begin(115200);
    while (!Serial);

    tmcSerial.begin(115200, SERIAL_8N1, SERIAL_RX, SERIAL_TX);
    while (!tmcSerial);

    pinMode(STEP_PIN_1, OUTPUT);
    pinMode(DIR_PIN_1, OUTPUT);
    pinMode(STEP_PIN_3, OUTPUT);
    pinMode(DIR_PIN_3, OUTPUT);
    
    pinMode(EN_PIN, OUTPUT);
    digitalWrite(EN_PIN, LOW);

    pinMode(SERVO_PWR_PIN, OUTPUT);
    digitalWrite(SERVO_PWR_PIN, HIGH);

    auto version1 = drv1.version();
    if (version1 != 0x21) {
        Serial.println("Driver 1 version mismatch!");
        Serial.println("Expected version 0x21, got: " + String(version1, HEX));
        fatalError();
    }
    auto version3 = drv3.version();
    if (version3 != 0x21) {
        Serial.println("Driver 3 version mismatch!");
        Serial.println("Expected version 0x21, got: " + String(version3, HEX));
        fatalError();
    }

    drv1.toff(5); // Enable driver
    drv1.rms_current(400); // Lower RMS current (example: 400mA)
    drv1.mstep_reg_select(true); // Enable microsteps via UART
    drv1.microsteps(16); 
    drv1.push();

    drv3.toff(5);
    drv3.rms_current(400);
    drv1.mstep_reg_select(true); // Enable microsteps via UART
    drv3.microsteps(16);
    drv3.push();
    
    // Configure stepper 1
    stepper1.setMaxSpeed(1000); // Set maximum speed (steps per second)
    stepper1.setAcceleration(500); // Set acceleration (steps per second^2)
    stepper1.moveTo(3200); // Move one full revolution (example: 200 steps/rev * 16 microsteps)
    
    while (stepper1.distanceToGo() != 0) {
        stepper1.run(); // Run stepper 1
    }

    // Configure stepper 3
    stepper3.setMaxSpeed(1000); // Set maximum speed (steps per second)
    stepper3.setAcceleration(500); // Set acceleration (steps per second^2)
    stepper3.moveTo(3200); // Move one full revolution (example: 200 steps/rev * 16 microsteps)

    while (stepper3.distanceToGo() != 0) {
        stepper3.run(); // Run stepper 3
    }

    servo.attach(SERVO_PIN); // Attach servo motor to pin 21

    for (int i = 0; i < 3; i++) { // Repeat the movement three times
        servo.write(90); // Move to 90 degrees (open position)
        delay(1000);     // Wait for ~1 second to ensure movement completes

        // Test closing functionality
        servo.write(0); // Move to 0 degrees (closed position)
        delay(1000);    // Wait for ~1 second to ensure movement completes
    }

    servo.detach(); // Stop sending signals to the servo
    
}

void loop()
{
}
```

Wiring (servo motor is from z-min pin, power is from hotbead and converted to 5v with mini360):

[image]

analogic, 2025-06-06:
https://github.com/analogic/mural - Here is fork with settings implemented - it is working but not yet fully tested.