nikivanov/mural #49: Belt slipping

adisharmagit, 2025-06-29:
Hello All,

First post and first project with 3d printed parts.

Since I borrowed 3d printer, i got the files printed on PLA. I see an issue where the belt slips when hung. Without weight, when the mural is in hand, the belt runs fine. When the belt slips, it seems like the gear is also stuck (i am just hearing motor clicks.)

This is my first time using a stepper motor. I see the motor is extremely quiet and gets very hot. Is it expected?

Since the recommended motor in the bom was not available, i am using the following motor. 

https://www.amazon.com/dp/B07TNPCWGK?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1
Usongshine Nema 17 Stepper Motor Bipolar Step Motor for Titan Extruder 3D Printer 4.1V 1A 13Ncm

Thanks again for the awesome project, i built it with my 6-year-old son, who is extremely excited to see it work real-time.

nikivanov, 2025-07-04:
I've never used these motors but they are rated for 1A, and Mural runs at 0.8A, so it should be ok. Did you set your VRef correctly to 1.1V like in the video?

The power supply should be ok at 3A

adisharmagit, 2025-07-10:
Yes, the ref voltage is set for 1.1v i also found the power brick was not supplying as written causing mural to stop. I switched to new anker 45W power supply which keeps it running but I still see belts slips. Wondering what others are using for motors as twotree motors are not available anymore.

adisharmagit, 2025-07-11:
Thanks for checking, I finally gave up replaced the right stepper motor, which was mainly clicking, and it works now. I also believe my previous power supply was not actually supplying the 3A current at 12V and thus causing other issues. 

Thanks again for the awesome project! I had fun building this with my son

nikivanov, 2025-07-11:
Great! I'll close the issue then.

closed 2025-07-11 by nikivanov